fossasia/pslab-documentation #3: Improve Repo structure

mariobehling, 2018-05-03:
* Merged Experiments.md into Readme.md
* List all experiments in Readme.md
* Move all Experiments in folder `/experiments`
* Move all screenshots into folder `/images`
* Move all schematics into folder `schematics`

mariobehling, 2018-05-14:
Move all screenshots into folder /images

The images have been moved to experiments/images instead.

harsh-2711, 2018-05-14:
Sure @mariobehling . I will do that

harsh-2711, 2018-05-14:
But images folder consists of three sub folders and if we shift it to /images rather than /experiments/images than we have to also change the code for displaying schematics and screenshots..

abhinavraj23, 2018-05-14:
@mariobehling I have originally done it as it was the way the docs were kept at [pslab-desktop-app](https://github.com/fossasia/pslab-desktop-apps/tree/development/docs/_apps) repo

mariobehling, 2018-05-14:
> But images folder consists of three sub folders and if we shift it to /images rather than /experiments/images than we have to also change the code for displaying schematics and screenshots

Yes, it needs to be changed to follow standard web practices.

PR #9 (2018-05-17): fix: Images folder moved to parent folder from experiments folder
Fixes: #3 #10
closed 2018-05-17 by mariobehling