fossasia/pslab-documentation #38: Add PR and issue template for this repository

AakashMallik, 2019-08-20:
PR and issue template missing from this repo.

Should be added to make it easier for developers to raise issues and PRs.

priyanshunayan, 2019-08-22:
@AakashMallik  PR template can be added as a file but adding issue template as a file is considered a legacy workflow. How can I add the issue template alternatively as I don't have the access to the Settings->Features->Issues->set up template?

AakashMallik, 2019-08-22:
> @AakashMallik PR template can be added as a file but adding issue template as a file is considered a legacy workflow. How can I add the issue template alternatively as I don't have the access to the Settings->Features->Issues->set up template?

Curious why adding issue file is considered legacy. Need some supporting doc to agree upon that right now.

priyanshunayan, 2019-08-22:
@AakashMallik I read this thing here. [Link](https://help.github.com/en/articles/manually-creating-a-single-issue-template-for-your-repository)

AakashMallik, 2019-08-22:
@priyanshunayan That doc states that this workflow is legacy as this will allow you to create a **single** issue template file for the whole repo. And for creating **multiple** issue template, they suggest using the new workflow. Do we need multiple issue templates right now? The answer is no. So the template file method should work just fine for now. 

:-)

AakashMallik, 2019-08-22:
Take reference from the PSLab Desktop repo. We have a PR template and Issue template in place. make needful modifications and copy paste..

priyanshunayan, 2019-08-22:
@AakashMallik Got it. Thank you so much. :)  I am onto it.

priyanshunayan, 2019-08-22:
@AakashMallik  I have added the files but when I opened the issue template of pslab-dektop, i was seeing this message. 
[image]

CloudyPadmal, 2019-08-23:
Yes, there's a new workflow where the first three options can be set by GitHub itself. https://github.com/fossasia/neurolab-android/issues/new/choose

AakashMallik, 2019-08-23:
> Yes, there's a new workflow where the first three options can be set by GitHub itself. https://github.com/fossasia/neurolab-android/issues/new/choose

@CloudyPadmal Should we switch to the new workflow??

closed 2019-08-23 by AakashMallik

CloudyPadmal, 2019-08-23:
Sure we can. https://github.com/fossasia/pslab-documentation/pull/41 check this one and suggest changes if anything's needed

PR #40 (2019-08-23): Add issue template and pull request template
Fixes #38 
Issue: PR and issue templates were missing from this repo.

I have added an issue template and a PR template in .github folder.

To verify you can check in .github folder in the root of the repository.