fossasia/pslab-documentation #48: Create Sphinx documentation in gh-pages branch

mariobehling, 2019-10-22:
Build a documentation website using CNAME docs.pslab.io in gh-pages branch.

priyanshunayan, 2019-11-06:
@mariobehling can I work on this issue?

mariobehling, 2019-11-07:
No need to ask for issues that are open for some time and where is no activity. Just go ahead and make a PR when you are ready.

kumuditha-udayanga, 2020-01-10:
I'm working on it  🙂

kumuditha-udayanga, 2020-01-15:
@mariobehling sir the issue is been fixed please review :)

mariobehling, 2020-01-21:
The next step is to make the Travis build work and to deploy the site in the gh-pages branch.
[image]

orangecms, 2020-08-13:
Deployment is now in place, so closing this.

closed 2020-08-13 by orangecms