fossasia/pslab-documentation #51: Autobuilds currently failing Config value: 'site_dir'. Error: The 'site_dir' should not be within the 'docs_dir'

mariobehling, 2019-10-31:
ERROR   -  Config value: 'site_dir'. Error: The 'site_dir' should not be within the 'docs_dir' as this leads to the build directory being copied into itself and duplicate nested files in the 'site_dir'.(site_dir: '/home/docs/checkouts/readthedocs.org/user_builds/fossasia-pslab-docs/checkouts/latest/_build/html', docs_dir: '/home/docs/checkouts/readthedocs.org/user_builds/fossasia-pslab-docs/checkouts/latest') 

Aborted with 1 Configuration Errors!

mariobehling, 2019-10-31:
Still failing

ERROR   -  Config value: 'site_dir'. Error: The 'site_dir' should not be within the 'docs_dir' as this leads to the build directory being copied into itself and duplicate nested files in the 'site_dir'.(site_dir: '/home/docs/checkouts/readthedocs.org/user_builds/fossasia-pslab-docs/checkouts/latest/_build/html', docs_dir: '/home/docs/checkouts/readthedocs.org/user_builds/fossasia-pslab-docs/checkouts/latest') 

Aborted with 1 Configuration Errors!

orangecms, 2020-08-13:
obsolete now - closing thusly

closed 2020-08-13 by orangecms