fossasia/pslab-documentation #62: PSLab Experiment: Finding directions using phone.

Devansh3712, 2020-01-05:
**Describe the issue**

I am conducting an experiment to find directions using PSLab android app.

**Would you like to work on the issue?**

Yes, I would love to work on the issue.

>>>The Experiment

AIM: To find directions using PSLab in an android phone.

MATERIALS REQUIRED: Android Phone with PSLab installed and having compass sensor.

PROCEDURE: 
1. First of all install PSLab from Playstore in your android phone and open the application.
2. After opening the app, search for Compass instrument in the app and open it.
3. On opening the instrument, you'll see a dial with 4 vertices, namely North (N), South (S), East (E), West (W).
4. Rotate your device 360 degrees to check whether it is working or not and calibrate it. The instrument also has sub vertices for South East (SE), North East (NE), South West (SW) and North West (NW).
5. Now you can check the directions using this instrument by only your android phone!

OBSERVATIONS:
1. The PSLab application shows the right direction on being rotated.
2. The application is accurate in showing directions.
3. Really easy to use and a perfect interface for android.

RESULT:
Therefore, we can find directions from android phone using the PSLab's Compass intrument.

[image]
[image]
[image]

NishithP2004, 2020-01-05:
There is tutorial already added for this issue and therefore this issue may be closed.

NishithP2004, 2020-01-05:
Duplicate of #60

orangecms, 2020-08-16:
Agreed to this being a duplicate, closing thusly.

closed 2020-08-16 by orangecms