fossasia/pslab-documentation #114: update collaboration/contribution guideline towards markdown

orangecms, 2020-08-16:
We have an older section that encourages ReST, though we wanted to use markdown; only document headlines (the first one on top) really need to be written in ReST now as far as I understand the Sphinx docs. I'm not sure about remarks (`| Warning: Be sure to read this carefully |`).

The file in question is `collaborate/Readme.md`.

orangecms, 2020-08-30:
Addendum:

It seems like remarks are actually just empty tables with only a headline; hence the line after it `| --- |`.

I.e.,
```
| Remarkable headline |
| --- |
```

looks like a "hint"; and here on GitHub:

| Remarkable headline |
| --- |

pizofreude, 2023-01-21:
From what I understand about this issue are:
1. An update for the collaborate/Readme.md file to serve as guideline for any user to start collaborating or contributing in the [Pocket Science Lab Project](https://pslab.io/)
2. The guideline must be written to guide collaborators using Markdown instead of ReST format for their project description and tutorials.
3. Provide open-source alternative tool that tailored towards Markdown instead of using [notex](https://www.notex.ch) or any other ReST editor.
4. Explain Best-Practices workflow for all user to refer when contributing.
5. Only document headlines (the first one on top) really need to be written in ReST
6. For remarks: place empty tables with only headline in this Readme.md as follows:
| Remarkable headline |
| --- |

Am I understanding the issue correctly? I would love to contribute and become member of the FOSSASIA community. Feel free to assign me this task. Thanks.

Satarupa22-SD, 2024-09-05:
@orangecms could you please assign this issue to me. Thanks!

orangecms, 2024-09-05:
There is no assignment, just file pull requests.

Satarupa22-SD, 2024-09-06:
@orangecms I have created a PR(#132) on the same, please look into it. I am open to any feedback or changes as needed. Thanks!