fossasia/pslab-documentation #120: PS Lab Logo missing

anilmaurya61, 2020-08-24:
Screenshot for better understand

[image]

nikhanal, 2020-08-25:
Can I work on this issue?

bessman, 2020-08-25:
Sure! [This logo](https://github.com/fossasia/pslab-android/blob/development/app/src/main/res/drawable/logo.png) should be displayed there.

nikhanal, 2020-08-25:
I am new to open source projects can you me guide me?

bessman, 2020-08-25:
Sure. The documentation is written in Sphinx. The [readme](https://github.com/fossasia/pslab-documentation/blob/master/Readme.md) explains how to build the documentation website. The layout of the generated website is defined in [conf.py](https://github.com/fossasia/pslab-documentation/blob/master/conf.py). If you look at [Sphinx's documentation](https://www.sphinx-doc.org/en/master/usage/configuration.html) you should be able to figure out how to add the logo.

nikhanal, 2020-08-25:
Is this ok?
[image]

bessman, 2020-08-25:
Yes, that looks good! Please open a pull request.

anilmaurya61, 2020-08-25:
> Is this ok?
> [image]

anilmaurya61, 2020-08-25:
> Is this ok?
> [image]

Yes! Please open PR with preview link.

swapnilsparsh, 2020-08-30:
Can I work on this issue ?

bessman, 2020-08-30:
From the looks of it @nikhanal has already fixed it. However, they haven't opened a pull request yet. @nikhanal, please do so as soon as possible.

@swapnilsparsh, let's wait until after the weekend. If no PR has been opened by then, feel free to start working on this issue.

nikhanal, 2020-08-30:
I am really sorry that I was late.
I have opened the pull request please view it.

orangecms, 2020-08-30:
Done in #121 - thank you a lot @nikhanal :)

closed 2020-08-30 by orangecms

nikhanal, 2020-08-30:
> Done in #121 - thank you a lot @nikhanal :)

My pleasure