fossasia/pslab-documentation #180: About Page Redesign using Astra + Gunternberg

Nimeshan, 2025-06-27:
I am planning to redesign the About Page, and the Pages relevant to it, and I am also planning to remove the Team page and add the content from it to About Page itself, to keep the Website Simple.

Nimeshan, 2025-06-28:
Today's Update with About Page

[image]

Nimeshan, 2025-06-29:
[image]

[image]

Update with the Team Section in the About Page, and Home Page

Nimeshan, 2025-07-04:
@CloudyPadmal I need @bessman's Image and Short Content for it.

bessman, 2025-07-05:
Yep!